wiedehopf/tar1090 #387: Lag on IOS and Web browser.

nycmadu, 2025-08-30:
Hello, Current issue where Tar1090 is running over 10k planes with NO SDR but ran by a Global API [API](https://global-adsb.chrisnyc.net/stream

[image]

)

OS Ubuntu 24.04 (AWS EC2)

Installed packages: nginx, nodejs 20, pm2

1. Wrote custom server.js (Node) to:
2. Connect to https://global-adsb.chrisnyc.net/stream (SSE)
3. Output site/data/aircraft.json + site/data/receiver.json

Problems that has came up was list below

1. Tar1090 UI becomes extremely laggy with ~10k–12k aircraft loaded.
2. On iOS Safari it often freezes/crashes.
3. On desktop  memory usage climbs several GB (4 GB max) and becomes unstable. 

my pc has ddr5 64gb ram so that should be a issue and is ran on server side so stress should be put on the server.

wiedehopf, 2025-08-30:
You're not using the correct mode used by all the aggregators that use it witih readsb API.

Worst lag is from the table not being limited probably.
If globeIndex is in receiver.json the behaviour will differ significantly.
But that also means it will request different data.

gvanem, 2025-09-01:
> Wrote custom server.js (Node) to:
> Connect to https://global-adsb.chrisnyc.net/stream (SSE)

Out of interest, how do you query for this data?
When trying with curl here, all I get is a `403 Forbidden`. Any docs for this API?

nycmadu, 2025-09-01:
@gvanem looks like you got blocked by CloudFlare for the 403 I've whitelist you so I would try again and yes, Airplanes.live

mpentler, 2025-09-04:
> [@wiedehopf](https://github.com/wiedehopf) I will look into it. Thank you!

https://wiki.oarc.uk/flight:adsb-server-writeup This guide I wrote may help.